digidotcom/xbee-micropython #1: unable to connect to AWS

venkyfun, 2018-08-08:
host = b'a28s4ep4ew56xw'  # ex: b'a1p3gcs127hy79'
region = b'us-west-2'  # ex: b'us-east-2'
thing_name = b'IMEI_0889'  #ex: b'IMEI_63890'

aws_endpoint = b'%s.iot.%s.amazonaws.com' % (host, region)
ssl_params = {'keyfile': "cert/aws.key",
              'certfile': "cert/aws.crt",
              'ca_certs': "cert/aws.ca"}  # ssl certs

ERROR:
connecting...
Traceback (most recent call last):
  File "<stdin>", line 48, in <module>
  File "<stdin>", line 37, in publish_test
  File "/flash/lib/umqtt/simple.py", line 72, in connect
OSError: [Errno 7006] ENXIO

tomlogic, 2018-08-09:
The `ENXIO` is a DNS lookup failure.  Are you sure you're connected (`ATAI==0`) before calling into this code?

tonymor9026, 2019-05-16:
Hello I'm having the same issue, I am trying to run the simple.py with

** mqtt = MQTTClient("XBee", "asl564f7xgf0p.iot.us-west-2.amazonws.com", ssl=True, ssl_params={"keyfile": "cert/aws.key", "certfile": "cert/aws.crt", "ca_certs": "cert/aws.ca"})
>>> mqtt.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/lib/umqtt/simple.py", line 57, in connect
OSError: [Errno 7006] ENXIO
>>> **

I've been able to connect and receive data running the aws-http.py

I read that depending on the CA I have to remove the -ats from the Endpoint

webdeveloperninja, 2020-06-09:
Any luck with this? I am also stuck. Docs are not super clear on which Root ca to use

mikewadsten, 2020-06-09:
@webdeveloperninja Our documentation here should help clear up which CA to use: https://www.digi.com/resources/documentation/digidocs/90002219/Default.htm#tasks/t_install_certificates.htm

mikewadsten, 2023-02-16:
@aiwilson I believe that part of the issue you're running into is that you are using the wrong TLS certificate. Per https://www.digi.com/resources/documentation/digidocs/90002219/Default.htm#tasks/t_install_certificates.htm you need to use the Starfield Services Root Certificate Authority - G2 certificate, _not_ the Starfield Class 2 file. Can you try that and report on whether you succeed?

aiwilson, 2023-02-20:
@mikewadsten I was able to connect using your recommendations. However, now when I try to connect using the MQTT library, I get the following errors:

>>> from umqtt.simple import MQTTClient
>>> mqtt = MQTTClient("XBee", "arg85w9dsr008-ats.iot.us-east-2.amazonaws.com", ssl=True, ssl_params={"keyfile": "cert/aws.key", "certfile": "cert/aws.crt", "ca_certs": "cert/aws.ca"})
>>> mqtt.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/lib/umqtt/simple.py", line 108, in connect
IndexError: bytes index out of range

I have the following policy attached to the IoT Thing/Certificate:

mikewadsten, 2023-02-20:
@aiwilson When you used the aws_https sample, did you also have THING_NAME set to "XBee"? AWS IoT depends on both the TLS client certificate you offer up, as well as your client ID, to identify the device. If the client ID you're using does not match the "thing", I believe AWS just immediately kills the connection, which would explain the IndexError you see (since the `.read(4)` call will return empty in that case).

aiwilson, 2023-02-20:
When I run the aws_https sample I did have it match the IoT Thing name (eSwitch). It connects, but never gets any data:
 +-----------------------------------+
 | XBee MicroPython AWS HTTPS Sample |
 +-----------------------------------+

- Waiting for the module to be connected to the cellular network... [OK]
- Connecting to AWS... [OK]
- Sending shadow request for thing 'b'eSwitch''... [OK]
- Waiting for data...

mikewadsten, 2023-02-20:
Reviewing your device policy, I think you're missing the `iot:Connect` action.

aiwilson, 2023-02-20:
That was it. Thank you!