digidotcom/xbee-micropython #2: HTTPS JSON problems

Loynis2018, 2018-09-11:
i can connect with http with json files but when i try to connect with https the micrphyton stop when i write variable.json() and i need boot the chip. does know someone what happend?

tomlogic, 2018-09-12:
Can you share the code you're using that works, along with the code that doesn't work?  Does the server you're connecting to support https connections?

Loynis2018, 2018-09-13:
the code is simple, i have one sever with http and other with https, with http works good, but when i used https the chip blocks itself.

code:
import urequests
a = urequests.get("https:\\pageweb.com\") (i cant show the page beacuse it is firm web test)
i received the 200, how ok, but if i use json block chip.
in explorer https works and show json.

tomlogic, 2019-03-19:
In your pasted code, you're using a backslash in the URL.  Have you tried `a = urequests.get("https://pageweb.com")`?

tomlogic, 2019-03-19:
If you want someone to look into this, please re-open this issue and paste the output of a terminal session demonstrating the failure, and hopefully with a URL that we can use to reproduce the failure.

closed 2019-03-19 by tomlogic