digidotcom/xbee-micropython #3: Connection refused

vasilyvs, 2018-11-13:
I cannot connect to AWS from my DIGI XBee modem
I made all just like this: https://github.com/digidotcom/xbee-micropython/tree/master/samples/cellular/aws
I am having this result when I try to run "aws_https.py":

network connected
connecting...
Traceback (most recent call last):
  File "<stdin>", line 37, in <module>
  File "<stdin>", line 24, in https_test
OSError: [Errno 7111] ECONNREFUSED 
the line 24 is:     w.connect((hostname, 8443))

or for and "aws_publish.py":

Traceback (most recent call last):
  File "<stdin>", line 34, in <module>
  File "<stdin>", line 23, in publish_test
  File "/flash/lib/umqtt/simple.py", line 72, in connect
OSError: [Errno 7111] ECONNREFUSED
the line 72 is:         self.sock.connect((self.server, self.port))

AWS endpoint parameters:
host = b'a1eredb3oecuwt-ats' '
region = b'us-west-2'  
thing_name = b'C352613070085950D'  
hostname = aws_endpoint = b'%s.iot.%s.amazonaws.com' % (host, region)

Thing, Policy and Certificates were created and activated just now, root certificate I have got from here: https://www.symantec.com/content/en/us/enterprise/verisign/roots/VeriSign-Class%203-Public-Primary-Certification-Authority-G5.pem. 
Open policy from here: https://github.com/digidotcom/xbee-micropython/blob/master/samples/cellular/aws/policy-open.json. My module is XBCLTE Cat1 Verizon IM# 352613070085950, firmware version 100C. I work a long time with WiFi modules connected to this AWS account without problems. I cannot understand what I missed, help me please...

mikewadsten, 2018-11-13:
Since your `host` variable ends in `-ats`, it seems like you're trying to connect to the endpoint set up to use Amazon's root certificate as opposed to a Verisign certificate. This could definitely explain getting ECONNREFUSED. https://aws.amazon.com/blogs/iot/aws-iot-core-ats-endpoints/

Try removing the `-ats` from the `host` variable. If that doesn't work, then try using one of the Amazon root CAs listed here: https://docs.aws.amazon.com/iot/latest/developerguide/managing-device-certs.html

vasilyvs, 2018-11-14:
Thanks for the help! It was lucky to make connect after remove "-ats" from the host

closed 2018-11-14 by vasilyvs