digidotcom/xbee-micropython #6: XBee3 Zigbee I2C Issue

JacoBezuidenhout, 2019-03-18:
Good day,

I love the repo thank you. 

One issue is that I tested this in an XBee Lte-M module and ordered 20 XB3 Zigbee modules just to find that they do not support I2C. Do you perhaps have any workarounds?

tomlogic, 2019-03-19:
MicroPython I2C support is in the next XBee3 Zigbee firmware release, expected by the end of April.

closed 2019-03-19 by tomlogic

JacoBezuidenhout, 2019-03-25:
Thank you for the update. Do you know when it will be added to the digimesh firmware? The XBee3 digimesh firmware doesn't even support MicroPython but Digi advertises it...

JacoBezuidenhout, 2019-03-25:
We are busy with a project POC that requires it. Do you think it is possible to get access to Beta firmware? If we get this project we will be deploying 1000+ xbees.

tomlogic, 2019-03-25:
Have you been in touch with anyone in Digi International's sales department?  They might be able to give you an official answer on MicroPython for DigiMesh and I2C support, along with an early release of the software.  If you don't have a contact at Digi yet, email me (look at the git commit history for my address) and I'll pass your message on to someone there.