digidotcom/xbee-micropython #14: ImportError: cannot import name relay

tou2, 2019-07-15:
I am trying to build and upload "relay frame led" example on pycharm but it gives this error:

ImportError: cannot import name relay

Note: Almost all the other examples are working fine
I am using Xbee3 digimesh pcb antenna.

DigiMChh, 2019-07-15:
I believe the User Relay frame feature is only available on the XBee3 ZigBee firmware: https://www.digi.com/resources/documentation/Digidocs/90002219/#container/cont_send_receive_user_data.htm

This page was incorrect in stating that it is supported on 802.15.4 and DigiMesh: https://github.com/digidotcom/xbee-micropython/tree/master/samples/xbee/communication/relay_frames_button

tomlogic, 2019-07-15:
There should be a DigiMesh release later this month that includes support for User Relay frames.

802.15.4 support should be out by the end of September.

tou2, 2019-07-16:
Thanks, I will test it once its released.

closed 2019-07-16 by tou2