digidotcom/xbee-micropython #18: Hey Please help I cant find xbee.py file on my /WEB/python directory. How can I fix this?

Flowmot1on, 2019-10-22:
My xbee industrial gateway give this error every reboot;
Traceback (most recent call last):

Oct 22 11:57:41 local7.info pylog:   File "xbgw_main.py", line 13, in 

Oct 22 11:57:41 local7.info pylog:     from xbgw.xbee.manager import XBeeEventManager

Oct 22 11:57:41 local7.info pylog:   File "/home/builder/bamboo-agent-home/xml-data/build-dir/XBGW-MASTER4-JOB1/xbgw/xbee/manager.py", line 1, in 

Oct 22 11:57:41 local7.info pylog:   File "/userfs/WEB/python/xbee.py", line 20, in 

Oct 22 11:57:41 local7.info pylog: AttributeError: 'module' object has no attribute 'AF_ZIGBEE'

rubenmoral, 2019-10-22:
Hi @Flowmot1on,

That issue is not related to this project, please contact tech.support@digi.com for further guidance.

Best regards.

closed 2019-10-22 by rubenmoral

Flowmot1on, 2019-10-23:
hi @rubenmoral 

  Support services is not responding. I just need xbee.py script for userfs/WEB/python directory.
This script special for industrial version. I can't run my python script.

rubenmoral, 2019-10-23:
Hi @Flowmot1on, please give them a couple of days to respond. If not, let me know and I will escalate the problem.