digidotcom/xbee-micropython #34: UART not found

tuorpete, 2020-03-13:
When trying "uart_echo" sample Micropython gives an error "ImportError: cannot import name UART".
I looked into machine-module and it indeed does not have UART in it. 
What seems to be the problem?

mikewadsten, 2020-03-13:
What model of XBee are you using? Per the Digi MicroPython Programming Guide, only XBee Cellular and XBee 3 Cellular support `machine.UART` at this time: https://www.digi.com/resources/documentation/digidocs/90002219/default.htm#reference/r_features.htm%3FTocPath%3D_____2

tuorpete, 2020-03-14:
I see. I use the zigbee variant. I did not know what secondary uart meant but it seemed to me like a basic thing that would be available in all devices right from the start :(

closed 2020-03-14 by tuorpete