digidotcom/xbee-micropython #37: ZHA sensor example

Ulrar, 2020-04-11:
Hi Digi,

I've been trying to figure out how to use an xbee to publish data to an existing ZHA network coordinator for a few months now, but the doc doesn't really seem to have anything on that subject.
The digi forum hasn't helped unfortunately, before giving up and using some esp8266 instead, any chance to get an example of a simple ZHA sensor in this repo (or anywhere else) ?
I'm basically just trying to read load cells and send the reading to zigbee2mqtt / homeassistant.

While trying to figure it out myself I've had a few people get back because they wanted to do the same, so I think that'd be a very beneficial example, I'm not the only one failing at this.

Thanks !

zackyancey, 2020-04-15:
If all you need is to send some ZCL commands, `xbee.transmit()` accepts arguments for cluster ID, profile, etc. (The arguments are documented [here](https://www.digi.com/resources/documentation/digidocs/90002219/default.htm#reference/r_function_transmit.htm)). I'm not familiar with zigbee2mqtt, but if sending your data to a specific endpoint/cluster on the coordinator is enough `xbee.transmit()` can do that.

If you need full support for a server to discover/enumerate the module, then you need to enable [ZDO passthrough](https://www.digi.com/resources/documentation/digidocs/90001539/#reference/r_cmd_ao.htm) (set **AO** to 7) and handle all of the requests sent by the server. ZDO passthrough to MicroPython isn't supported in the currently released firmware, but it is going to be added in version 100A which is releasing at the end of April. Supporting server discovery will be possible in version 100A, but you'll need to parse and respond to all of the server's commands in MicroPython. An example of how to do this would be useful, as those commands can be hard to find documentation on. Hopefully we can get an example like that soon, once it's fully supported by the firmware.

Ulrar, 2020-04-15:
Hi,

Zigbee2mqtt is just a coordinator, when a new device connects to the network it will send discovery requests and enumerate the endpoints, and then select from the list of [supported devices](https://www.zigbee2mqtt.io/information/supported_devices.html) (philips hue, xiaomi, ikea .. any of the usual manufacturers of zigbee enabled devices) how to use each endpoint depending on the announced model, id, vendor, description ..

So basically yes, it needs to support server discovery. I didn't realize that wasn't currently possible, but that certainly explains why I haven't been able to make it work, I was wondering why xbee was answering requests on it's own regardless of the mode I was in.

Glad to hear it'll be possible at the end of April ! For now I just have one sensor reading to publish so I'll just hardcode everything around it when that new firmware gets released and see how it goes.

Examples would be amazing, it has been hard to find much about all of this online, thanks for your answer ! :)

altaic, 2020-05-16:
Thanks for the useful information. An example would be incredibly helpful. I'm currently working my way through the 869 page (!) ZCL spec, along with a bunch of forum threads with bits of information.

My goal is to get a simple proof of concept working with an xbee3 and a couple sensors connected to it via I2C, and have it report/publish the sensor data to an existing ZHA network (Home Assistant, in my case). The second step will be to handle commands to switch stuff on and off.

So far, I've got the sensors over I2C working in micropython, and I've managed to get the xbee3 connected to Home Assistant, in which the device is shown to have five sensors (analog_input) and 15 switches (on_off). I'm not sure what those all correspond to, however I'm trying to set it up to just report two sensors (and nothing else) right now: temperature and humidity. I gather that will require setting `AO=7` and writing an event handler in micropython to respond to incoming ZDO (or ZCL?) commands.

Any pointers on that last part (the micropython event handler) would be really appreciated!

altaic, 2020-05-19:
I found @Ulrar's [StackOverflow post](https://stackoverflow.com/questions/60465782/zigbee-interview-process-xbee), which was very useful. In particular, the two links (Ulrar's and tomlogic's):
* [Connecting an Xbee 24ZB to Smartthings Hub](http://web.archive.org/web/20180218010916/https://www.falco.co.nz/electronic-projects/xbee-to-smartthings/)
* [XBee ANSI C Library](https://github.com/digidotcom/xbee_ansic_library)

The first is a nice introduction to the inner-workings of Zigbee, while the second has an implementation of ZCL that's far easier to follow than the ZCL spec. There are two directories which are especially relevant: [include/zigbee](https://github.com/digidotcom/xbee_ansic_library/tree/master/include/zigbee) and [src/zigbee](https://github.com/digidotcom/xbee_ansic_library/tree/master/src/zigbee). I've begun sketching out a ZCL library for micropython based on that code, and later on I may extend it using the ZCL spec.

One thing to note, though: a full python implementation may be too large to fit on the XBee in text form, so it'll likely have to be compiled to `pyc` or `pyo` byte-code.

waytotheweb, 2021-05-09:
I was wondering if you'd made any progress on this at all to save some time having to go through something similar?

As has been mentioned, a fully working ZHA example would really be a boost for those of us wanting to make end-devices for home automation. I have something cobbled together that work for me for my specific needs, but it isn't pretty and isn't using Zigbee ZCL with proper Cluster ID's  and endpoints, as it should.

Ulrar, 2021-05-09:
I personally gave up, trying to get it to work in micropython I always ran
into space issues just trying to get basic ZCL implemented, not even
including the logic I actually want on top of it.

I ended up going with an esp chip over wifi, at least until someone comes
up with a way to fit enough of the ZCL on xbee.

waytotheweb, 2021-05-09:
Funnily enough, once I started working on this I have also moved on to use the ESP32 with very low power options. It would, as you say, be a real advantage for home automation to be able to use the XBee3's to create small Zigbee projects that talk ZCL with HA hubs.

chbarg, 2021-05-09:
I also gave up some time ago. There is not much support from Digi unlike the great support that you can find on other Zigbee platforms.
I have played with https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/ which let's you create a zigbee2mqtt compatible module. 
If you find any good information, please post.
Take care

Shulyaka, 2021-09-15:
A standard XBee ZHA quirk will give you access to XBee's internal ADCs. I doubt you can implement a whole new ZCL cluster but you can create a virtual template sensor just in HA on top of the standard quirk if your sensor is analog, no micropython needed.