digidotcom/xbee-micropython #38: XBee MicroPython Not Found in PyCharm Marketplace

mtm12, 2020-04-16:
I'm trying to load the XBee MicroPython plugin for PyCharm. The XBee MicroPython plugin is not found in the PyCharm marketplace. If I try to load the plugin from disk from xbee-micropython-1.0.3.jar, I get the error: Plugin 'XBee MicroPython' is incompatible with this installation
PyCharm version: 
PyCharm 2020.1 (Professional Edition)
Build #PY-201.6668.115, built on April 7, 2020
Licensed to Marc McLean
You have a perpetual fallback license for this version
Subscription is active until April 16, 2021
Runtime version: 11.0.6+8-b765.25 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 10 10.0
GC: ParNew, ConcurrentMarkSweep
Memory: 1979M
Cores: 16

rubenmoral, 2020-04-17:
Hi @mtm12,

The plugin is not compatible with PyCharm 2020.1 yet, that's why you don't see it on the Marketplace. We are working on supporting it, we expect to have a new version of the plugin in the next two weeks.

In the meantime, you can install the previous version (2019.3) from the JetBrains site if you want to use the plugin: https://www.jetbrains.com/es-es/pycharm/download/other.html.

Sorry for the inconvenience.

mtm12, 2020-04-17:
Hello @rubenmoral ,
OK. Thank you,
Marc

closed 2020-04-17 by mtm12