digidotcom/xbee-micropython #41: uasyncio and uselect?

altaic, 2020-05-19:
I was planning to use the `uasyncio` micropython module to write a robust event handler for #37, but the XBee3 micropython does not include it. So, I copied it over from the [official micropython-lib repo](https://github.com/micropython/micropython-lib) as described in [this guide](https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md#01-installing-uasyncio-on-bare-metal).

However, `uasyncio` is uses the `uselect` module, which the XBee3 micropython also lacks, and unfortunately this one needs to be compiled into the micropython runtime.

Is there some simple work-around? Or otherwise can the `uselect` module be added to the firmware in the future?

mikewadsten, 2020-05-20:
Hi @altaic,

As is mentioned in the typehints for the `uselect` module, we only support `uselect` on XBee and XBee 3 Cellular devices with firmware ending in x15 or newer: https://github.com/digidotcom/xbee-micropython/blob/master/typehints/common/uselect.pyi#L188

I'm not sure what the current plans are for implementing `uselect` on the XBee 3 RF products. I will ask around.

altaic, 2020-05-20:
Thanks, @mikewadsten. I had missed that note in the `uselect` typehints. Having `uselect` would be really helpful.

altaic, 2020-06-08:
@mikewadsten

mikewadsten, 2020-06-08:
@altaic No updates at this time. I've submitted a request internally to add support but I can't make any guarantees on when/if it will happen.

altaic, 2020-06-15:
Thanks; please let me know when there’s news.

altaic, 2020-07-06:
It’s been a while, and I’d like to restart the projects that are blocked by this issue. Any updates?