digidotcom/xbee-micropython #42: aws_publish IPPROTO_SEC type error

StevenSOConnell, 2020-05-29:
Using the aws_publish example with umqtt installed on XBC KTE-M/NB-IoT modem firmware 11415 with L0.0.00.00.05.08 installed.  When I run the example I get:

File "<stdin>", line 58, in <module>
File "<stdin>", line 47, in publish_test
File "/flash/lib/umqtt/simple.py", line 61, in connect
TypeError: must use IPPROTO_SEC

What does this mean?  MicroPython version 3.4.0.

In the Digi MicroPython Programming Guide PDF page 135 it says:

- You can only wrap a socket created with protocol IPPROTO_SEC.  Python3 uses IPPROTO_TCP

Does this mean this example will *not* run on my setup using Python 3?

mikewadsten, 2020-05-29:
Hi @StevenSOConnell ,

Based on that error message, it looks like you might not be using the `umqtt` library from this repository (https://github.com/digidotcom/xbee-micropython/tree/master/lib/umqtt). The error coming from line 61 is a sign you are using the `umqtt` library from `micropython-lib`: https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L61

We would recommend using the Digi XBee MicroPython PyCharm Plugin to run these samples and develop MicroPython applications for the XBee. http://cms.digi.com/resources/documentation/digidocs/90002219/#container/cont_pycharm.htm

StevenSOConnell, 2020-05-29:
Thanks for the quick response.

Earlier I installed the PyCharm plugin & PyCharm and got the exact same error message, even using the PyCharm example project.  I down-graded the XBee firmware from 11415 to 11410 and everything started working in PyCharm.  Why do I have to do this?

Thanks,

We would recommend using the Digi XBee MicroPython PyCharm Plugin to run these samples and develop MicroPython applications for the XBee. http://cms.digi.com/resources/documentation/digidocs/90002219/#container/cont_pycharm.htm . Running this sample using our PyCharm plugin will automatically use the correct version of umqtt for the XBee.

mikewadsten, 2020-05-29:
@StevenSOConnell What if you use the 11413 firmware?

Also try formatting the XBee filesystem and retesting on 11415. It's possible the `micropython-lib` `umqtt` library was still there and was used when you tried it in PyCharm the second time.

StevenSOConnell, 2020-06-01:
I downgraded to 11412 and got it to work.

I didn’t try 11413 yet.