digidotcom/xbee-micropython #47: XBee 3 Cellular LTE-M/NB-IoT - memory leak

kmodar, 2020-07-10:
Hello,

I am running a small micropython code loop on XBee 3 Cellular LTE-M/NB-IoT hardware, firmware version 11415.  Using examples from the digi documentation and GitHub I simply instantiate a XBee and a network object, test network connection, calculate my next wake up time and then force sleep.  Things are running fine except that before I sleep I look at my free and allocated memory.  Then I run garbage collection and then look at the memory again.  I'm noticing a small memory leak.  Its not every time through the loop but over time it adds up.  My solution as of now is to set a threshold level and safely shutdown the cell module and reboot. Just wondering if anyone can help debug where I am going astray.  Attached is a txt file of my code.

[main_py.txt](https://github.com/digidotcom/xbee-micropython/files/4903926/main_py.txt)

Thanks,
Kris

kmodar, 2020-07-10:
Sorry, I forgot to attach the test data.
[Digi_memory_testing.xlsx](https://github.com/digidotcom/xbee-micropython/files/4903945/Digi_memory_testing.xlsx)

Thanks,
Kris

mikewadsten, 2020-07-10:
The MicroPython garbage collector is very conservative, so a small "memory leak" while in an execution loop isn't really concerning.

Do note that all the calls to e.g. `print(gc.mem_alloc())` will, themselves, trigger a memory allocation.

Have you tried testing what happens if you replace the sleeping logic (everything from `'''<30s just wait; >10m something wrong'''` down) with something like `time.sleep_ms(50)`? That way you can see if the GC truly is failing to collect the extra memory. (You know this is the case if you hit a `MemoryError`.)

kmodar, 2020-07-13:
Latest update:

I have attached my latest code, which is a bit cleaner. As you suggested, I replaced the x.sleep_now(x, False) with just a time.sleep_ms(x). Where x = 120000.  This seems to work without leaking a single byte.  If I only swap out the time.sleep_ms(x) with the x.sleep_now(x, False) I once again start dropping bytes every few times through the loop.

Thanks,
Kris

[main_py.txt](https://github.com/digidotcom/xbee-micropython/files/4913402/main_py.txt)