digidotcom/xbee-micropython #50: Cannot install xbee micropython plugin for pycharm

iftahgi, 2020-09-07:
[image]

The xbee micropython plugin cannot be find on pycharm, and when trying to install it from disk, the error message above appears.

Thank you

rubenmoral, 2020-09-07:
Hi @iftahgi,

Thanks for reporting this issue. We are already working on giving support for PyCharm 2020.2 to the plugin, we expect to release a new version in the next days. We'll let you know when it is available. In the meantime, you can still use the plugin with PyCharm 2020.1 or lower.

Sorry for the inconvenience.

iftahgi, 2020-09-07:
Many thanks for this prompt reply!

rubenmoral, 2020-09-17:
Hey @iftahgi, just to let you know that the new version with support for PyCharm 2020.2 is now available.

https://plugins.jetbrains.com/plugin/12445-xbee-micropython

closed 2020-09-17 by rubenmoral