digidotcom/xbee-micropython #67: ImportError: no module named 'UART

mohamadnoor991, 2021-04-19:
Hello, i have Xbee3 on the XBee grove board, when I am trying to execute some code I have an error about "ImportError: no module named 'UART".
How I can use new modules on Xbee3 ?
Should I upload these modules and how?
i need a quick answer please to start the execute my code. 
Thanks

SWoto, 2021-04-27:
It's probably due to XBee3 having no secondary UART interface, only primary UART, so you have limitations.

[image]

You probably need to go here:
[image]

[Digi MicroPython](https://www.digi.com/resources/documentation/digidocs/PDFs/90002219.pdf)

mohamadnoor991, 2021-04-28:
@SWoto 
Thank you, my friend yes this is the problem.
but if you have any idea about how I can connect the **xbee3 module on XBee grove THT** with **ESP32** to communicate and send data between them? 
Thanks for your answer.

SWoto, 2021-04-28:
You can use the side pins, this way you will be able to use normal jumpers to connect XBee's primary UART to ESP32 UART.
[image]

The connection need is:
```
02 DOUT -> ESP32 RX
03 DIN  -> ESP32 TX
```

You'll be able to send messages with something like the code below:
```python
from sys import stdin, stdout
import time

def receive_data():
    data = stdin.buffer.readline()
    return data

def write_data(data_out):
    stdout.write(bytearray(data_out))

data_out = 'Hello World from Cellular'
complement_data_in = 'XBee Received from ESP32: '

time.sleep(1)
write_data(data_out)

while True:
    data = receive_data()
    if data is not None:
        time.sleep(.1)
        write_data(complement_data_in+data.decode('acii'))
        time.sleep(1)
```

Also, avoid using USB and primaryUART together, I'd some problems with it because USB data gets send to my other board and my other board data goest to the computer too, so it gets messy.