digidotcom/xbee-micropython #68: Question: XBee + Serial Comunication

SWoto, 2021-04-27:
Hello everyone,

I've going through XBee's documentarion regarding micropython and its two modules, the XB3 and the XBee Cellular. My goal is to implement a Modbus protocol on top of the RS485, we only need RX/TX pins for serial comunication, but I've some questions....

For what I understood, it seems pretty simple to use UART comunication with the XBee Cellular given that it has two UARTs. But, does anyone knows if it is possible to implement serial comunication between XB3 and another device/sensor? Looks like it has some limitation related to the  primary UART.

Shulyaka, 2023-07-16:
It is possible to communicate over primary UART, but please note that it will always send this text after each reboot:
```
Loading /flash/main.mpy...
Running bytecode...
```

tomlogic, 2023-07-16:
The [gps_uart](../../tree/master/samples/gps_uart) sample demonstrates communication with another device via the secondary serial port, which leaves the primary port for MicroPython debugging, or usage of API frames with a host processor.