digidotcom/xbee-micropython #81: Source code for network.py

webdeveloperninja, 2022-04-17:
I am troubleshooting a transient bug. Seeing the source code for network.py would be helpful. I only see the network.pyi interface file. Is there a link to the source code to network.py?

Thanks

webdeveloperninja, 2022-04-17:
This is the code with the issue. https://github.com/webdeveloperninja/SiemensBeltScaleIntegrator/blob/master/telemetry_client.py

When I send telemetry I get [Errno 7107] ENOTCONN. I have a guard clause checking `self.connection.isconnected()`. I still get the error. This happens after about 20 hours of transmitting every 10 minutes.

closed 2022-04-17 by webdeveloperninja

mikewadsten, 2022-04-18:
I see you closed this issue, so maybe you got past the problem, but in any case, here are a few answers for you.

There is no `network.py` source code to share for two reasons: 1) Digi intellectual property/copyright reasons, and 2) the `network` module is actually written in C code and integrated directly into the XBee firmware.

As for the ENOTCONN error - that most likely is coming from a socket send or recv call, and doesn't indicate that the _XBee_ is _disconnected_ from the cellular network, but rather that the TCP socket has been closed, most likely by the cellular carrier as a result of inactivity or even just their own rules about how long-lived TCP sessions may be. I believe that if you make use of the `keepalive` argument to the MQTTClient constructor, you may be able to mitigate this, by causing the server to send some traffic more often (say, every 5 minutes).

webdeveloperninja, 2022-04-18:
This helps to clarify things. I was sleeping the device and closing the socket. I believe I have it fixed. I am going to run the device for a couple weeks, so far so good. Thanks for response.

mikewadsten, 2022-04-18:
Yeah, if the XBee is put to sleep, all open sockets are torn down, and so you would have to recreate your MQTT client (or at least do something like call .sock.close and then .connect again - you won't be able to just call .disconnect because the write call there will fail) after each sleep cycle.