digidotcom/xbee-micropython #98: discover module not defined in micropython for Xbee.discover

robomike, 2025-01-24:
There is a micropython sample for Xbee.discover to find nodes on the network:

[ Xbee.discover.](https://github.com/digidotcom/xbee-micropython/blob/98a58d1284082f3a50f03dae8b68673450be5858/samples/xbee/network_discovery/main.py#L31) 

However, discover is not a module defined in micropython Xbee.

Also, the Digi website says this is available

https://www.digi.com/resources/documentation/Digidocs/90002219/#reference/r_example_xbee.discover.htm?TocPath=XBee%2520device%2520examples%257C_____4

If the python module is unavailable, how can we access this information in micropython?

mikewadsten, 2025-01-24:
Which model of XBee are you using? The `xbee.discover` function only exists on XBee 3 Zigbee, 802.15.4 and DigiMesh.

robomike, 2025-01-24:
XBee 3 Global 

[image]

mikewadsten, 2025-01-31:
If you have multiple XBee 3 Cellular modules, and they are joined to the same cellular network, and your network carrier allows direct communication between devices in that way (for example, if you are using a private APN configured as such), then sure, one XBee can communicate with other XBees -- but you'll need to address them directly. Otherwise, no. Mobile-terminated IP traffic, even within the same network or on the same cell, is almost always blocked as a matter of course on cellular networks.

This is why we don't implement ATND or `xbee.discover()` on Cellular modules.

robomike, 2025-01-31:
Hi, I have an Xbee3 Global LTE and it needs to communicate to an Xbee module:  https://www.sparkfun.com/xbee-3-module-pcb-antenna.html  

Our plan is to use the Xbee3 Global as a hub, to what would be a mesh network of embedded Xbee3 modules.

mikewadsten, 2025-01-31:
The type of product you are looking for is a [gateway](https://www.digi.com/products/embedded-systems/digi-xbee/digi-xbee-gateways). XBee 3 Cellular devices are not gateway devices. XBee 3 Cellular does not implement 802.15.4 (communication over Zigbee, DigiMesh, or 802.15.4 is not possible).

I would advise you to reach out to Digi's sales team for help with your use case.